makerbase-mks/MKS-Robin-Nano-Firmware #1: always doing restart

fenoloji, 2020-03-01:
I compiled the code. But always doing restart. How can i solve?

mks-viva, 2020-03-01:
Hi, do you mean restart specifically for motherboards?
Our compilation tutorial needs to be improved.

fenoloji, 2020-03-01:
Yes, motherbord restart.

Also your code "https://github.com/makerbase-mks/MKS-Robin-Nano-Firmware/blob/Nano-firmware-v2.x.x/EWARM/mksRobinNano/Exe/Robin_nano35.bin"  restarting motherboard.

  But "https://github.com/makerbase-mks/MKS-Robin-Nano/blob/master/Nano%20firmware%20v2.x.x/Mks_Nano35_V2.0.0%20Release%20file.rar" in Robin_nano35.bin don't restart motherboard.

mks-viva, 2020-03-01:
I probably know what caused the problem. It may be that the compiled bin file is not transcoded. I will update this tutorial tomorrow. Please wait for my good news. 
I sent you the binEncode_setup.exe installation file. You can use the software to transcode the compiled bin file, and then copy the transcoded bin file to the TF card and insert it into the motherboard for upgrade. Note the bin file name: Robin_nano35.bin

fenoloji, 2020-03-02:
Hİ, Thank you for your answer. When will you send binEncode_setup.exe to me?

makerbase-mks, 2020-03-03:
[image]
Is it not sent successfully? If so, don't worry. I have updated the tutorial on how to compile the firmware, you can refer to: https://github.com/makerbase-mks/MKS-Robin-Nano-Firmware/wiki/How-to-build-update-nano-firmware

fenoloji, 2020-03-03:
Hi, all done. Everthings works. You are wonderful.  I want to prepare the Turkish language file and change the code. Would you allow?

makerbase-mks, 2020-03-04:
You can certainly modify, MKS open source is to serve more enthusiasts.

closed 2020-03-04 by makerbase-mks

fenoloji, 2020-03-08:
hi,  can you help me?
how can i change the contents of the GBK16.bin
Example;

   orjinal  [cid:78d7c1ee-134f-465e-813e-566255822427] [cid:c7a790ca-3abc-47fc-b41b-fd23eff7b0e1]
                                                                                                                        The shape I want.
________________________________
Gönderen: mks-viva <notifications@github.com>
Gönderildi: 1 Mart 2020 Pazar 16:24
Kime: makerbase-mks/MKS-Robin-Nano-Firmware <MKS-Robin-Nano-Firmware@noreply.github.com>
Bilgi: fenoloji <fenoloji@hotmail.com>; Author <author@noreply.github.com>
Konu: Re: [makerbase-mks/MKS-Robin-Nano-Firmware] always doing restart (#1)

I probably know what caused the problem. It may be that the compiled bin file is not transcoded. I will update this tutorial tomorrow. Please wait for my good news.
I sent you the binEncode_setup.exe installation file. You can use the software to transcode the compiled bin file, and then copy the transcoded bin file to the TF card and insert it into the motherboard for upgrade. Note the bin file name: Robin_nano35.bin

makerbase-mks, 2020-03-08:
Hi, this is a Chinese character font, you may be not use it. Of course, I can also consult software engineers to see if they can help you solve this problem.

fenoloji, 2020-08-28:
> I think I have the same problem! My display is always restarting. I have encoded it with the software provided!

ok. my problem solved.