makerbase-mks/MKS-Robin-Nano-Firmware #2: Problem opening Project file

Paul-F-Hanson, 2020-03-07:
Having followed the given instructions, My First attempt to load the project produces a dialog box with the following message
The project 'mksRobinNano' contains the unknown tool 'CMSISDAP_ID'.

Once open performing rebuild all yields the following messages ( I've simply copied them all, sorry if this is plain wrong to do).

Building configuration: mksRobinNano - mksRobinNano 
Updating build tree... 
 
52  file(s) deleted. 
Updating build tree... 
G26_Mesh_Validation_Tool.cpp  
Warning[Pa050]: non-native end of line sequence detected (this diagnostic is only issued once) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\Third_Party\Marlin\G26_Mesh_Validation_Tool.cpp 1 
Fatal Error[Pe035]: #error directive: "No E stepper plug left for Z2!" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\Third_Party\Marlin\pins.h 673 
Error while running C/C++ Compiler 
Helvetica26.c  
Warning[Pa050]: non-native end of line sequence detected (this diagnostic is only issued once) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\User\ui\Multi_language\Helvetica26.c 1 
Fatal Error[Pe1696]: cannot open source file "GUI.h" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\User\ui\Multi_language\Helvetica26.c 28 
            searched: "C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\User\ui\Multi_language\" 
            searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 6.4 Evaluation\arm\inc\" 
            searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 6.4 Evaluation\arm\inc\c\" 
Error while running C/C++ Compiler 
Helvetica36.c  
Warning[Pa050]: non-native end of line sequence detected (this diagnostic is only issued once) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\User\ui\Multi_language\Helvetica36.c 1 
Fatal Error[Pe1696]: cannot open source file "GUI.h" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\User\ui\Multi_language\Helvetica36.c 28 
            searched: "C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\User\ui\Multi_language\" 
            searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 6.4 Evaluation\arm\inc\" 
            searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 6.4 Evaluation\arm\inc\c\" 
Error while running C/C++ Compiler 
M100_Free_Mem_Chk.cpp  
Warning[Pa050]: non-native end of line sequence detected (this diagnostic is only issued once) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\Third_Party\Marlin\M100_Free_Mem_Chk.cpp 1 
Fatal Error[Pe035]: #error directive: "No E stepper plug left for Z2!" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\Third_Party\Marlin\pins.h 673 
Error while running C/C++ Compiler 
MarlinSerial.cpp  
Warning[Pa050]: non-native end of line sequence detected (this diagnostic is only issued once) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\Third_Party\Marlin\MarlinSerial.cpp 1 
Fatal Error[Pe035]: #error directive: "No E stepper plug left for Z2!" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\Third_Party\Marlin\pins.h 673 
Error while running C/C++ Compiler 
Marlin_main.cpp  
Warning[Pa050]: non-native end of line sequence detected (this diagnostic is only issued once) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\Third_Party\Marlin\Marlin_main.cpp 1 
Fatal Error[Pe035]: #error directive: "No E stepper plug left for Z2!" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\Third_Party\Marlin\pins.h 673 
Error while running C/C++ Compiler 
Print.cpp  
Warning[Pa050]: non-native end of line sequence detected (this diagnostic is only issued once) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.cpp 1 
Error[Pe020]: identifier "class" is undefined C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Printable.h 25 
Error[Pe020]: identifier "class" is undefined C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Printable.h 33 
Error[Pe065]: expected a ";" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Printable.h 34 
Warning[Pe012]: parsing restarts here after previous syntax error C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Printable.h 37 
Error[Pe020]: identifier "class" is undefined C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 36 
Error[Pe065]: expected a ";" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 37 
Warning[Pe012]: parsing restarts here after previous syntax error C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 58 
Error[Pe147]: declaration is incompatible with "__thumb __softfp size_t print(char const *)" (declared at line 59) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 60 
Error[Pe018]: expected a ")" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 61 
Error[Pe147]: declaration is incompatible with "__thumb __softfp size_t print(char)" (declared at line 60) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 61 
Error[Pe018]: expected a ")" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 62 
Error[Pe147]: declaration is incompatible with "__thumb __softfp size_t print(unsigned char, int)" (declared at line 61) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 62 
Error[Pe018]: expected a ")" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 63 
Error[Pe147]: declaration is incompatible with "__thumb __softfp size_t print(int, int)" (declared at line 62) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 63 
Error[Pe018]: expected a ")" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 64 
Error[Pe147]: declaration is incompatible with "__thumb __softfp size_t print(unsigned int, int)" (declared at line 63) C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Middlewares\arduino\stm32\cores\arduino\Print.h 64 
[... cut ...]
Fatal Error[Pe1696]: cannot open source file "stm32f10x_gpio.h" C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Src\wifi_upload.cpp 4 
            searched: "C:\Users\Ching\Downloads\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\MKS-Robin-Nano-Firmware-Nano-firmware-v2.x.x\Src\" 
            searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 6.4 Evaluation\arm\inc\" 
            searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 6.4 Evaluation\arm\inc\c\" 
Error while running C/C++ Compiler 
 
Total number of errors: 281 
Total number of warnings: 180

makerbase-mks, 2020-03-08:
Can you provide pictures, such as loading project, compiling project and so on

Paul-F-Hanson, 2020-03-09:
All I did was follow the links on this page

    Get firmware bin file
    To compile

I registered the IDE opened the project ans attempted compilation, I did nothing else.
Perhaps the system from which the tutorial was developed had other  prerequisites?
Try following the guidance on a machine not previously used for the purpose and you should see the same.
I am however using Windows 7 but there was no mention of using a more recent OS only.

makerbase-mks, 2020-03-13:
Maybe the version of IAR software is different. The version of my IAR software is 7.1.0. I can compile this code very well.

Hazmit, 2020-03-23:
I'm having a bit of trouble compiling as well since I'm familiar with using the Arduino software for compiling marlin directly. 

Would it be possible for you to compile the nano firmware for me with the pins for extruder 0 thermistor switched with the extruder 1 thermistor? 

I have a board with a bad port for E0 temp sensor and want to flash it to look at E1 temp sensor instead so I can use the board and just move the connection.

Paul-F-Hanson, 2020-03-23:
Hello Hazmit.
The Arduino IDEisn't capable ofperforming the compilation.  You will need Microsoft Visual Studio Code and thePlatform IO plugin.
It's not something I've tried myself but you will find some guidence on Youtube. The same board is shipped with a new Core xy printer from China.
https://escope.de/posts/sapphire-pro-marlin/
[https://escope.de/static/hp_sapphire-1-13a62701910c6559317ba6dc26b1eb03-c6823.jpg1zB]<https://escope.de/posts/sapphire-pro-marlin/>
TwoTrees Sapphire Pro and Marlin 2.0 - electronic-telescope<https://escope.de/posts/sapphire-pro-marlin/>
The Sapphire Pro from TwoTrees comes shipped with a MKS Robin Nano board. While the board is open, its firmware is not and so you can hardly modify anything. Instead of replacing the board I managed to get Marlin working this machine.
escope.de

Good luck.