makerbase-mks/MKS-Robin-Nano-Firmware #3: turn off butonu?

fenoloji, 2020-03-21:
Why is there no button to turn off the machine on the screen?

karohan, 2020-03-22:
Modül satıyorlar onun için.

fenoloji, 2020-03-22:
> Modül satıyorlar onun için.
Selamlar, sebep o değil. Biraz düşününce anladım. Ekrandan kapatsak bile açarken bir anahtara ihtiyaç olacak. Ekran kapalı olduğunda nereden açacağız makineyi. Arkadaki kapatma anahtarı zaten açık. oradan açamayız. Demek ki bunun için bir tetikleme anahtarı var. Onu da beraberinde satıyorlar.
    Ekrana kapatma butonu koyulabilir. Ancak açmak için bir tetikleme anahtarına yine ihtiyaç olacak.

karohan, 2020-03-22:
Türkçe karakter seti ekleyebildin mi ?

fenoloji, 2020-03-22:
> Türkçe karakter seti ekleyebildin mi ?

Türkçe dil seçeneğini ekledim. Şu anda kendi bilgisayarımda derledim. Çalışıyor. Ancak githuba yüklemedim henüz. Türkçe karakterler bir acayip çıkıyordu. Onları bile tek tek düzelttim. a harfini el yazısındaki gibi yaptım, ı harfi ve ç harfini de düzelttim.
[image]

karohan, 2020-03-22:
Çok özel değilse paylaşabilir misin? Türkçe karakter sorunu yaşıyorumda

fenoloji, 2020-03-22:
> Çok özel değilse paylaşabilir misin? Türkçe karakter sorunu yaşıyorumda

Bitmeden paylaşamam. Çalışmamı proje yöneticisine göndericem. Ondan eklemesini isteyeceğim.

karohan, 2020-03-22:
Peki yinede teşekkür ederim. https://www.facebook.com/groups/1544993445653957/ Grubumuza beklerim ;)

closed 2020-06-18 by fenoloji

JuculianD, 2020-08-25:
Hello? I have the same problem. How did you solve???