makerbase-mks/MKS-Robin-Nano-Firmware #5: Original BL-Touch not working on nano V1.2

olaf-h, 2020-04-07:
I try to connect a BL-Touch 3,1 to the supplied ports on the nano board either as a z-Limit switch or only for bed leveling on z-max (MKS Firmware 2.0.1). The sensor is not working in any configuration (with marlin firmware it's running) When I switch to a BL-Touch Clone (Touch 3d) it's working properly with the original MKS Firmware. Is there any configuration I need to adjust or is there simply an HW issue or a bug in the original MKS Firmware

makerbase-mks, 2020-04-07:
Hi! Nano firmware is compatible with hardware prior to BLTOUCH V3.1, but BLTOUCH V3.1 modifies the hardware so that existing firmware cannot drive it. Subsequent firmware versions, we are compatible with BLTOUCH V3.1

JinwooJang, 2020-04-11:
Dear makerbase
could you let me know plan to release next version ?

the reason not working BLTOUCH V3.1 is this PWM duty cycle ?
as I see, out of range PWM output which is required by BLTOUCH v3.1.

> 
> 
> Hi! Nano firmware is compatible with hardware prior to BLTOUCH V3.1, but BLTOUCH V3.1 modifies the hardware so that existing firmware cannot drive it. Subsequent firmware versions, we are compatible with BLTOUCH V3.1

makerbase-mks, 2020-04-11:
It is caused by different pulse widths. We will deal with this problem in these two days.

olaf-h, 2020-04-15:
Great, thnx. With Marlin it’s working properly, but I like the original MKS software :)

makerbase-mks, 2020-08-26:
@olaf-h 
I will close this question temporarily.

closed 2020-08-26 by makerbase-mks