makerbase-mks/MKS-Robin-Nano-Firmware #6: not implement AUTO_BED_LEVELING_BILINEAR/AUTO_BED_LEVELING_UBL

ochkariky, 2020-04-07:
hello! Sorry for my bad English.
I`m use AUTO_BED_LEVELING_BILINEAR function on Sapphire Pro with 3DTouch.
firmware 2.0.1

G29 is work: 
SENDING:M420 V
Bilinear Leveling Grid:
      0      1      2      3      4
 0 +0.014 -0.253 -0.506 -0.717 -0.889
 1 +0.059 -0.199 -0.458 -0.693 -0.877
 2 +0.078 -0.163 -0.418 -0.659 -0.856
 3 +0.079 -0.160 -0.409 -0.638 -0.826
 4 +0.088 -0.147 -0.402 -0.626 -0.806
echo:Bed Leveling On
echo:Fade Height 10.00

But, this map not implement for move G0/G1 - movements is planar. Z is constant.
Does anyone have 3DTouch | BLTouch?
i`m try UBL and G29 P1 etc...  - The result is the same. Mesh is complete but not inplement for Z movement.

My config [robin_nano35_cfg.txt](https://github.com/makerbase-mks/MKS-Robin-Nano-Firmware/files/4446718/robin_nano35_cfg.txt)

ochkariky, 2020-04-09:
strange. Maybe the eeprom error was. On compiled firmware 2.0.1 - AUTO_BED_LEVELING_UBL works !
By the way, why are the firmware 2.0.1 sources not uploaded? there is a have BabyStep in it during printing

MKS-Sean, 2020-04-13:
https://github.com/makerbase-mks/MKS-Robin-Nano/tree/master/Nano%20firmware%20v2.x.x

MKS-Sean, 2020-04-13:
This problem has been fixed in the V2.0.2 version, and the source code has also been updated

ochkariky, 2020-04-14:
Thanks for the new version and source code!
I tried setting the auto level in version 2.0.2 many times. Now, after calibration, motor Z moves when moving along x and y. But the result of level calibration with the parameter BED_LEVELING_METHOD == AUTO_BED_LEVELING_BILINEAR is unpredictable. Twice I drove a nozzle into the table (I printed the first slow test with the G30 command and gives an error of 0.05-0.1mm at the calibration points.
I perform the auto level with this sequence in the gcode file:
G28;
G0 Z5 f3000;
G29 J;
G29;
 and print test rectangles in layers from 0.2-0.15-0.1mm.

Returned to using the mode:
 BED_LEVELING_METHOD == AUTO_BED_LEVELING_UBL in the same firmware 2.0.2.
I use the code:
G28; Home
G0 Z5 F3000; ++ up 5mm for move
G29 P0; UBL reset
G29 P1; UBL autolevel
G29 A; UBL autolevel Activate
M420 S1; autolevel ON
After that, the first 0.1mm layer prints perfectly. re-checking with the G30 gives an error of not more than 0.03mm. The main drawback of this mode is the incorrect interpolation beyond the borders where BLTouch cannot reach - for me this is a wide 40mm band