makerbase-mks/MKS-Robin-Nano-Firmware #12: Can I build TFT32 version?

Spider84, 2020-06-17:
Can I build TFT 3.2 version?
VALUE_DEFAULT_X and VALUE_DEFAULT_Y constants are not defined :(

makerbase-mks, 2020-06-18:
I'm sorry, this version is currently unable to support 3.2 inches

Spider84, 2020-06-18:
Can I upgrade my 3.2 to 3.5?

makerbase-mks, 2020-06-19:
Robin TFT35 link: https://www.aliexpress.com/item/4000484271283.html?spm=2114.12010612.8148356.3.695c788eaOeRcU

Spider84, 2020-06-19:
This lot is more expensive than a controller with a display assembly.
Not funny... I can buy new one more nano with tft35...

makerbase-mks, 2020-08-26:
@Spider84 
1. Robin Nano35 bootloader: https://github.com/makerbase-mks/MKS-Robin-Nano/blob/master/Bootloader/Robin_nano35_boot_200608.hex
2. Use Jflash tools upload it

Spider84, 2020-08-26:
@makerbase-mks 
thx, But I already do that. But now I searching for TFT35 LCD :)