makerbase-mks/MKS-Robin-Nano-Firmware #19: Problemas con BlTouch

RobertoMarco, 2020-08-27:
Tengo instalado el Bltouch de Makerbase en la MKS Robin Nano V1. 2 y configurado, todo actúa correctamente pero el Bltouch no saca la aguja por lo que no realiza las medidas. Probado con Firmware V1 y V2,  con Marlin la pantalla se queda en negro. 
Me pueden ayudar por favor

Gorkag3d, 2020-09-10:
Hola!
Encantado de escribir en castellano sobre la MKS Robin Nano, con la BLTouch, yo la tengo montada con esa placa y funciona correctamente, yo uso el firmware "Mks_Nano35_V2.0.3.3 Release file", la tengo configurada y conectada en el ZMin ¿Como la tienes conectada? Zmin, Zmax ¿Normaly Open?

RobertoMarco, 2020-09-10:
Muchas Gracias, ya funciona bien conectado en Zmax

pistachin2, 2020-09-11:
Buenas, que gusto da ver que también se puede hablar en castellano, yo tengo otro tipo de problema,
A mi me funciona bien, pero el problema que tengo es que a cada autolevel que se realiza se suma el offset.
Es decir, que recién calibrado la impresión es perfecta, en el segundo autolevel es filamento queda apenas pegado a la cama, y en el tercero y sucesivos, no se pega a la cama, solo se hace una bola de filamento.
Como digo, la sensación que da es que se va sumando el offset en cada autolevel.
Le tengo cableado y configurado como en el vídeo de MKS, y he probado con el firmware v2.0.3.2, el 3.3 y el 3.4.
Me tiene muy perdido el tema.
Un saludo.

Gorkag3d, 2020-09-11:
Hola!
La verdad es que si que parece raro, esta parte se me escapa un poco a mi también. Por lo que he ido leyendo, parece que puede ser algún problema a la hora de grabar en la eeprom o similar, pero no tengo ni idea de cual o como arreglarlo.
Me imagino que habrás ido probando a reestablecer manualmente el offset para cada impresión y haciendo un autolevel para cada impresión.
Probaré este fin de semana a ver si me pasa a mi también.
A mi también me pasan cosas raras con el autolevel de la BLTouch, miraré a ver si pueden venir de algo relacionado con lo que te pasa a ti.
Yo el principal problema con el autolevel es que lo hace unas cuantas veces bien, pero de repente deja de hacer la malla de puntos del autolevel, toma el primer punto y ahí se queda, me suele funcionar desconectar el cable de 2 hilos, cambiarlo al Zmax, conectar un final de carrera normal en el Zmin, y después volver a conectar la BLtouch al Zmin.

También se me ocurre que puedes probar a hacer el autolevel introduciendo el GCODE manualmente, puedes mirar en:

https://marlinfw.org/docs/gcode/G028.html

Por lo que he visto, el autolevel hace: G28;G29;M500.

Yo tengo la BlTouch en el Zmin, y la udo se final de carrera y Autolevel. 
La verdad es que procuro imprimir sin Autolevel, nivelando manualmente, porque he puesto la placa en una impresora DIY y de momento el EjeZ me sufre un poco cuando imprimo con autolevel, aunque espero perfeccionarlo y disfrutar de las impresiones con Autolevel.

Seguiré probando cosas.

pistachin2, 2020-09-11:
Yo lo tengo conectado en Zmax, y final de carrera mecánico en Zmin. Solo lo uso para el autolovel, no para sustituir el final de carrera de Z (porque la ubicación del BlTouch en homing esta fuera de la cama)
En el "fileteador" (Cura) tengo que con cada impresión haga un G28 y luego G29, ya sabes que si la placa hace un homing (G28) se borra la maya del autolevel.
Yo veo que suma el offset de Z a la hora de arrancar la placa, yo uso pronterface, si enciendes la placa y justo le das a conectar (por USB y hay que hacerlo nada mas encenderla) ves todo el chorro de arranque de la placa y en las ultimas lineas (de memoria creo que la tercera o cuarta por el final) aparece el offset de Z que tiene almacenado en la eeprom, no el valor que esta configurado, de hay detecto que se va sumando.
Actualmente tengo des-habilitado el autolevel para que no me de guerra.
**Tu problema me ha pasado más de una vez,** y por las pruebas que realice, llegue a la conclusión que es porque los valores de offset de X e Y del BlTouch se van fuera de rango, es decir, que si realizara las medidas con los valores que tiene, la placa entiende que el carro se chocaría con la estructura o no podría llegar a la coordenada para hacer la prueba (por decirlo de alguna manera), yo lo solucione poniendo unos offset mas pequeños hasta que me diseñe un soporte que estuviera lo mas pegado al extrusor, para que los valores fueran los mas pequeños posibles (tanto valores en positivo como en negativo).
También revisa que las conectores estén bien apretados, no es mala idea desplazar/doblar un poco, y con mucho cuidado, los pines para que no falsee la conexión (con mucho cuidado, muy poco y con la placa apagado)
No se si me explicado bien, si tienes alguna duda con el comentario me dices y te intento ayudar/explicar (hay veces que me explico como un libro cerrado, jeje)
Tengo abierto un ticket con este problema https://github.com/makerbase-mks/MKS-Robin-Nano/issues/63 a ver si me dicen algo la gente de MKS.
Gracias por tus comentarios.

Gorkag3d, 2020-09-11:
Muchas gracias Pistachin2,
Creo que me será de gran ayuda tu aportación, a ver si lo puedo probar este fin de semana y lo confirmo.
Había mirado para cambiar los puntos del autolevel en el "robin_nano35_cfg", pero si puedo arreglarlo desde el tft, me es más cómodo.
Ya veo que no te he aportado mucho y que pilotas bastante je,je.

Yo también utilizo "Cura" como "fileteador" con el plugin MKS Wifi para las miniaturas en el TFT.

Muchísimas gracias

pistachin2, 2020-09-13:
No es que pilote, me he pegado mucho con esta placa....
Una pregunta respecto al plugin "MKS Wifi", utilizas el que del Github o el del markerplace de Cura, te lo comento por que el del Github me hace bien las miniaturas pero me salen descolocados los botones para el movimiento, y el del markerplace me pasa al revés, me hace mal las miniaturas pero los botones me salen bien colocados.
Es por si solo me pasa a mi o es algo genérico.
Ya me contaras que tal las pruebas. 
Un saludo.

Gorkag3d, 2020-09-13:
Utilizo de Cura,
Me pasaba lo de las miniaturas, leí la solución, y hay que cambiar el tamaño de la imagen en la carpeta donde esté instalado el plugin de cura en el archivo:" SaveOutputDevice, la línea; save_file.write(utils.add_screenshot(image, 100,100, ";simage:")), el tamaño lo tendrás en 50x50 y hay que ponerlo mas grande.

Para encontrar la carpeta, en Cura\Ayuda (help)\Show configuration Folder\ Plugins\...

Suerte y me dices

pistachin2, 2020-09-22:
Muy buenas, he mirado lo que cometas del tamaño, lo tenia de serie en 100x100, he probado distintas configuraciones y siempre me aparece el logo "desplazado". seguiré usando el plugin de GitHub, aunque los botones me salgan superpuestos.
Me puedes pasar el enlace donde vistes la solución? por revisarlo si yo tengo alguna cosa más...
Muchas gracias de todas.

Gorkag3d, 2020-09-23:
Hola!
Te lo intento enviar, aunque no se muy bien como hacerlo, así que pruebo varios links, que van todos al mismo hilo.

Yo tengo la pantalla TFT35, no sé si tienes la misma, y los tamaños de las miniaturas son los mismos.

https://github.com/makerbase-mks/MKS-TFT/issues/177#issue-347615043

https://github.com/makerbase-mks/MKS-TFT/issues/177

[https://github.com/makerbase-mks/MKS-TFT/issues/177](url)

También he probado a imprimir usando el autolevel, y algo de que suma los offsets si que me pasa ¿Pudiste avanzar con ese problema?